rl-institut/smooth #2: There are no emissions in each components

LauWien, 2020-01-06:


j-ti, 2020-01-15:
emissions could be treated as
1. variable_emissions (independently from cost) which I do currently
or
2. emissions_cost (artificial cost) -> would need a factor to balance between emissions[kg] and cost [EUR]
which one should be realized? @LauWien

j-ti, 2020-01-16:
>     1. variable_emissions (independently from cost) which I do currently (commit [7d42326](https://github.com/rl-institut/smooth/commit/7d423269cb9cc8d49d5ef63aadfbac57e7c05f5f))

Together with Laura, we decided to keep it as variable_emissions [g/Wh] and, for now, keep the calculation in component.update_cost() next to the calculation of variable_costs [EUR/Wh]

j-ti, 2020-01-27:
commit 76efe6dbd52473c2fe905ac3c12f6b86c05ed511 
- emissions are now in [kg]
- variable name change cap_emissions to fix_emissions
- update emissions in a seperate function from cost

j-ti, 2020-03-02:
https://github.com/rl-institut/smooth/pull/15

closed 2020-03-02 by j-ti