rl-institut/smooth #7: MATLAB-haruspex function is missing in python

LauWien, 2020-01-06:


j-ti, 2020-01-16:
primary goals
- output timestep of error
- output states of components until timestep of error

j-ti, 2020-02-13:
[debugDataframe.txt](https://github.com/rl-institut/smooth/files/4197398/debugDataframe.txt)
[image]

j-ti, 2020-02-17:
currently debugging function will automatically be executed when optimization status is not ok and termination condition is not optimal and raise a runtime exception 
1. should it continue running when termination condition is not optimal but there are supoptimal flows generated?
2. should it be changed, so it will execute the debugging function only manually

closed 2020-04-07 by LauWien