rl-institut/smooth #50: Dev branch stopped working

stefansc1, 2020-03-25:
When executing run_smooth_example.py, the computation aborts with

``
ValueError: The parameter "power_max" is not part of the component
``

component_supply.py seems to be the culprit. Can be fixed by adding self.power_max = None. However, a new error stops the computation:

``
AssertionError (assert self.dependency_flow_costs is not None) from component.py, line 54
``

Expected behavior: run_smooth.py does not crash on dev branch.

j-ti, 2020-03-25:
should be solvable by adding a default attribute to supply and sink components, e.g. for supply:
 `self.dependency_flow_costs = 'flow: bel-->to_grid'`
otherwise it has to be inserted to the model.

b-r-u, 2020-03-25:
The `power_max`-part is fixed with https://github.com/rl-institut/smooth/pull/51/commits/b8b2259c0337a4ebe67910b92e986d60773a275c

j-ti, 2020-03-26:
Maybe add a clear assertion message, like: "Component XY has a variable cost, but no associated flow dependency AB is given."

b-r-u, 2020-03-26:
@j-ti This could be another PR.

ulrikeherrmann, 2020-03-27:
I pulled from the dev branch today and got the two error messages mentioned by @stefansc1 . Fixed it using the suggestions and now get a new error:
  File "C:\Users\Ulrike\python_project\smooth\smooth\framework\functions\update_fitted_cost.py", line 179, in get_dependant_value
    dependant_value = getattr(component, fitting_dict['dependant_value'][index])
TypeError: getattr(): attribute name must be string

ulrikeherrmann, 2020-03-27:
I also want to suggest that the dependency_flow_costs do not require to have a default value or to be inserted in the model definition for sink and supply components. Since self.name and self.bus_out / self.bus_in are defined in the model self.dependency_flow_costs can be obtained from 'flow: '+self.name+'-->'+self.bus_out for the supply or 'flow: '+self.bus_in+'-->'+self.name for the sink component.

b-r-u, 2020-03-27:
@ulrikeherrmann Could you take a look at this PR https://github.com/rl-institut/smooth/pull/51 and verify if it meets your criteria? The example is running without errors but maybe you see some semantic errors?

closed 2020-04-02 by timroepcke

j-ti, 2020-04-02:
> I also want to suggest that the dependency_flow_costs do not require to have a default value or to be inserted in the model definition for sink and supply components. Since self.name and [...]

@ulrikeherrmann Just for completeness: We agreed on a generic solution for variable costs and variable emissions, so as well the definition of dependency on a flow should be managed this way and not specific to supply or sink components. But we can keep this in mind, in case we change this fact

PR #61 (2020-04-08): Feature/optimization overhaul
### Reworked run_optimization

#### Floating point genes
Instead of binary values, floating point genes are used in individuals.

#### Keep the best
Instead of forgetting the best individuals, they are now selected and retained in the next generation (without double computation, of course).

#### Learn from the best
Instead of selecting random individuals, the best are used for crossover and mutation.

#### Repeatable
Same challenge in, same output.

#### Multi-objective
Optimize for cost, emission or weighted sum.

#### Less Third Party dependencies
Removed deap. [... cut]