rl-institut/smooth #95: Relicense under dual MIT / Apache v2

b-r-u, 2020-04-23:
We would like to relicense this repo to MIT / Apache v2.

To agree to relicensing, comment with :
```
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.
```

* [ ] @SpamAndEgg
* [ ] @b-r-u
* [ ] @ciaradunks
* [ ] @j-ti
* [ ] @jakobgemassmer
* [ ] @stefansc1
* [ ] @timroepcke
* [ ] @ulrikeherrmann

b-r-u, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

timroepcke, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

SpamAndEgg, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

ciaradunks, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

jakobgemassmer, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

j-ti, 2020-04-23:
I license past and future contributions under the dual MIT/Apache-2.0 license,
allowing licensees to chose either at their option.

PR #96 (2020-04-30): Change license to dual MIT/Apache-2.0
Resolves #95
closed 2020-04-30 by b-r-u