rl-institut/smooth #100: There is no battery electric vehicle

j-ti, 2020-05-05:


j-ti, 2020-05-11:
An electric vehicle can be modeled by:
- Sink (if only charging)
- Mobility Bus with Sink (Leaving) + Storage (Connected) + Source (Arriving)

Link: https://oemof.files.wordpress.com/2017/05/20170508_oemof_sector_coupling_notebook_schachler.pdf

nesnoj, 2021-02-05:
Hey @j-ti, I'm thinking of using this approach in another energy system model based upon pypsa-eur-sec. Has the approach in your example 32a0555 been used so far, is it working? If so, does it introduce a lot of complexity to the optimization problem? (due to varying battery capacity)

j-ti, 2021-02-05:
> Hey @j-ti, I'm thinking of using this approach in another energy system model based upon pypsa-eur-sec. Has the approach in your example [...]

Hey @nesnoj, I did not continue the approach and thus did not tested it very much, sill, I think it is a good way of implementing it with few complexity. In my opinion the varying capacity should not make a mathematical optimization much more complex, as they represent just additional constraints, if the timeseries for capacity, energy leaving and entering the system are fixed values.