rl-institut/smooth #147: the inital storage level is not the end storage level

LauWien, 2020-06-17:
This is important so the energy system doesn't get heat/hydrogen/electricity for free

j-ti, 2020-06-18:
As we optimize each timestep individually, it could happen that at the ultimate timestep we would have to (un-)fill the storage by a flow that exceeds the maximum flow.
My suggestion: We calculate the minimum flow needed to (un-)fill the storage and set it as a constraint.

j-ti, 2020-06-19:
If the storage needs to be equal at start and end of simulation => There must exist a variable sink and/or source for that energy carrier type to transport

j-ti, 2020-07-01:
[image]
[image]
[image]
[image]

j-ti, 2020-07-01:
H2 bus with and without incentives (ratio):
[image]
[image]

j-ti, 2020-07-01:
Electr. bus with and without incentives (rising VAC)
[image]
[image]

j-ti, 2020-07-02:
@ulrikeherrmann do you think the number of intervals to the end to balance storage level using the dynamic VAC (so the soft criterion) should be defined in the modeldescription or based on an auto scaled value using max. flow and capacity, like:
`if self.storage_capacity / (self.delta_max) > self.intervals_to_end:` 
which would represent the number of intervals to charge the whole storage, as an example.

j-ti, 2020-07-07:
electrolyzer can only be optimized by electrical VAC: Solution foreign states on supply:
    'fs_component_name': 'h2_storage',
    'fs_attribute_name': 'balance_ratio',
    'fs_threshold': 0.2,
    'fs_low_art_cost': 50,
    'fs_high_art_cost': -0.001,
=>
with soft constraint (foreign state of evolving balance) over all intervals + hard (min flow) constraint:
[image]
[image]

only with hard (min flow) constraint:
[image]
[image]

j-ti, 2020-07-07:
(3) hard constraint (min flow) + soft constraint (foreign state: balance only last **15 intervals**)