rl-institut/smooth #198: master branch is out-of-date

b-r-u, 2020-09-15:
The master branch is nearly a year old and does not serve any purpose. I propose merging the dev branch and continue development on the master-branch. At our current rate of development we don't have a useful stable baseline. If we want to support older versions of smooth in the future, we can create dedicated branches for that. 

Using the master branch we can also use the standard URL for readthedocs, ending with `latest`.

b-r-u, 2020-09-17:
That part about readthedocs was wrong. I just pointed the dev branch to `latest`:

https://smooth.readthedocs.io/en/latest/

j-brendel, 2021-09-23:
Master branch is now up to date with version 0.3.0

closed 2021-09-23 by j-brendel