rl-institut/smooth #212: MPC not fully reliable

j-brendel, 2020-11-23:
Without MPC: Battery from solar or from grid only:

[image]

[image]

Preview of possibility with MPC:
[image]

j-brendel, 2020-11-23:
To try out:
[UHAL_v2_MPC.zip](https://github.com/rl-institut/smooth/files/5584205/UHAL_v2_MPC.zip)

Use julian_working_branch of smooth

j-ti, 2020-12-01:
Grund, warum das Modell oben bei langem Zeithorizont mit linearem MPC nicht lösbar ist, liegt nach meiner Analyse genau an der Linearität die bei nahezuvollem Akku nicht angenommen werden kann. Genauer: Wenn E_max (charging) beim ersten Zeitschritt sehr sehr gering ist, weil die Batterie gerade voll ist, gilt der Wert in einer linearer MPC  für den gesamten Zeithorizont auch wenn zwischendurch wieder entladen wird.

Um auf den Vorteil von Linearität nicht verzichten zu müssen und es schnell umzusetzen, schlage ich vor:
1. die Batterie nicht maximal ausreizen (SOC_min, SOC_max) evtl. sowieso besser für die Lebensdauer
2. ein desired SOC einführen mit entsprechenden `artificial_costs`, so dass generell nicht bis SOC_max geladen wird, aber  weiterhin die Möglichkeit besteht, wenn es unbedingt nötig ist (auch besser für die Lebensdauer)

Ob das für dich so ausreicht müssen wir nochmal am Donnerstag besprechen

j-ti, 2020-12-01:
[image]

j-ti, 2020-12-10:
'battery_capacity_bt2': 300 * 1e3
'mpc_control_horizon': 30,

within battery component:
`self.p_in_max = (self.c_rate_charge * self.battery_capacity
            ) / self.efficiency_charge`
`self.p_out_max = (self.c_rate_discharge * self.battery_capacity)\
                         * self.efficiency_discharge`

[image]

j-ti, 2020-12-10:
same as above but:
'battery_capacity_bt2': 230 * 1e3

[image]

j-ti, 2020-12-10:
here again 'battery_capacity_bt2': 300 * 1e3
with the P max adaptation, the desired SOC with low_vac are not necessary:
switch default vac back to "keep Battery full (from grid) to minimize Bat":
`    'vac_in': -0.20 / 1000,
    'vac_out': 0.20 / 1000,`

[image]

j-ti, 2020-12-13:
example:
162 | li_battery |   | capacity | 56 |   | 50000 | **218721,02** | 250000 | 
163 | li_battery |   | capacity | 57 |   | 50000 | 50000        | 250000 |  

(SOC_i*(1-loss_rate)- SOC_min) * eff_decharge = flow(bat->bel)
(218721,02*(1-0,067/100/24)- 50000) * 0,95 = 160279,168336282

In reality the value output from oemof is 160279,17
As for oemof it is the end result it is seems no problem in which direction it rounds.

In this example which uses the battery down to SOC_min, one can see if smooth takes this rounded value the energy will not suffice to compensate for grid limitations by only smaller than 0.005. Same could appear in the other direction with SOC_max

j-ti, 2020-12-13:
As a test it is possible to add this value 0.005 Ah, which is very small in comparison to in this case E_min = 50 000 Ah
`self.soc = (df_storage[i_result][0]+0.005) / self.battery_capacity`
... then MPC works as expected:
I chose low feed_in tariff of 0.05, medium vac +/- 0.10 and the from grid cost of 0.181 per kW:

[image]

j-ti, 2020-12-13:
Thie intermediate results for the mpc for the battery are as follows:
[image]

j-ti, 2020-12-13:
Obviously, to add this value is a hack. Although as 0.005 << 50000 (example E_bat,min) and the price per timestep is very low (0.181 / 1000) * 0.005 = 9.05 x 10^-7
and in case timestep is 1 hour it becomes an error per Year of
(0.181 / 1000) * 0.005 * 24 * 365 = 0.0079278
There should be searched for a better way. For a timestep of 1s the price per year rises to 2.854008

j-ti, 2020-12-13:
As argued before the limitations for the battery's `self.p_in_max `and `self.p_out_max` as they were do not make sense as oemof already handles the constraints itself according to loss_rate, efficiency etc.

j-ti, 2020-12-13:
Full charging in one timestep is possible for c_rate=1 as p_max=battery_capacity is always bigger than battery_capacity - battery_capacity* SOC_min

j-ti, 2020-12-14:
> As a test it is possible to add this value 0.005 Ah, which is very small in comparison to in this case E_min = 50 000 Ah
> `self.soc = (df_storage[i_result][0]+0.005) / self.battery_capacity`
> ... then [...]

For similar reasons as described in [issue #212 comment](https://github.com/rl-institut/smooth/issues/218#issuecomment-744017840 ), the vac_in/out and feed in tariff should be chosen so that the negative cost of charing and discharging is not profitable when using solar energy, which has no variable cost.

I chose low feed_in tariff of -0.05 - 0.13 art_cost , medium vac +/- 0.10 and the from grid cost of 0.181 per kW:
This way charging and discharging does not appear between timesteps 125 - 140:

[image]
[image]