rl-institut/smooth #218: Battery Component doesn't work for MPC

j-brendel, 2020-12-10:
Because of the usage of niominal value in the P_max_in and P_max_out definitions, the battery can't be used for MPC Simulations.

Idea:
Silmlify  with Nominal Value for MPC compliance.

For Coherence updated Component should have these properties:

- Discharge and Charge Symmetrical
[image]
[image]

- keep 100 % possible
[image]

- Same behaviour 1 min and 60 min interval time

[image]
[image]

j-ti, 2020-12-13:
With the correct loss rate and efficientcy coefficients, eomof handels everything on its own and correctly.

The reason for the constant loading in #212 appears when -vac_in <= vac_out and -vac_in > variable grid cost, as it oemof does not distinguish between artificial or normal costs and makes profits from high inflow (negative cost) although outflow with equal vac as there is energy loss on in and out flow  (e. g. 100 Ah * 0.95 * 0.95 = 90.25 Ah). As energy flowing out is less, than inflow, even equal +/-vac results in a net negative value.

=> vac_in and vac_out should always be below the grid costs.

j-brendel, 2020-12-14:
With these changes, only one consistency criteroin is broken: 
Charging and discharging are not symmetrical:
[image]
[image]

Could we modify max_power in battery component to:

        self.p_in_max = self.c_rate_charge * self.battery_capacity / self.efficiency_charge
        self.p_out_max = self.c_rate_discharge * self.battery_capacity * self.efficiency_discharge

This solves this issue, do you think it creates others?

[image]
[image]

I used these files to create plots for consistency checks:
[example_Battery.zip](https://github.com/rl-institut/smooth/files/5690254/example_Battery.zip)

j-ti, 2021-01-11:
https://github.com/rl-institut/smooth/issues/212#issuecomment-742537429

j-ti, 2021-01-14:
PR  #223 was merged containing the new maximum nominal power definition

j-brendel, 2021-05-12:
Solved with #261

closed 2021-05-12 by j-brendel