rl-institut/smooth #265: apply investment mode

JassemKh, 2021-09-09:
hey everyone, 

I did some simple basic example with oemof.solph using the investment mode in the Flow() to determine the optimised energy dispatch and capacities of the components based the costs. Now I'm wondering wether it's possible to apply that in smooth for example to determine of optimised capacity of a fuel cell or an electrolyzer based on the costs. In the smooth documentation I see, the user should define the maximum power of the PEM electrolyzer, as nominale value. so my question it's possible to use the investment mode applying smooth without giving an nominale value ? 
 
I still be a beginner in the filed of energy modelling and Oemof. I would be grateful for any help and support.

j-brendel, 2021-09-09:
Hej JassemKh,

Thanks for your question! So far smooth is not designed to be used with the investment mode of oemof.solph.

In order to determine the optimized capacities of components using smooth there is a optimization tool. You can try the basic example here: https://github.com/rl-institut/smooth/blob/dev/smooth/examples/run_optimization_example.py
Here you have the option to give minimum and maximum values for each property that you want to optimize and as a result you will get the optimum value within this range according to the objective (e.g. annuity_total(costs) and annual_total_emissions as in the example).

I hope that answers your question?

Using the investment mode of oemof is an interesting option though. If you find a good way to include this into smooth or have an interesting idea we are very happy if you let us know.

Best Julian

JassemKh, 2021-09-10:
Hi Julian,

I understand, thank you so much for your reply!
yeah it would be nice to integrate such an option! as I said I still be a beginner but if something comes to mind I will let you know!

best Jassem